ahalekelly/Dettlaff #25: To Do: Build Web App

ahalekelly, 2023-01-21:


darkcheftar, 2023-01-25:
I want to Help you with this Issue, Can you please share more about this?

ahalekelly, 2023-01-25:
Thanks for the offer! We don't have a spec written out, but we need a simple web app to configure some electronics by reading and writing protobufs over the WebBluetooth API. How did you find this project? What's your level of web development experience? I see a React project on your profile?

darkcheftar, 2023-01-25:
I am a good enough web developer, I want to contribute to open source, Starting with your project.

darkcheftar, 2023-01-25:
I don't know anything about Nerf Modding but, I can research about it.
Currently researching about Protobuf

darkcheftar, 2023-02-01:
Hey @ahalekelly 
Please Let me know more about this

JsamsIvyTech, 2026-05-15:
Hey i want to help if you need it. Can I know more about what your wanting or the scope of the project.